adafruit/Adafruit_Keypad #1: Keypad angled presses causing misfires

jedgarpark, 2018-11-13:
Pressing buttons at an angle other then pretty straight down from the top often cause a pair or whole column of buttons to fire. In testing the megademo sequencer this can cause the sequence to stop playing accidentally, or even crash. @deanm1278 and @ladyada have both mentioned adding some filtering to fix this, which may be a conductive pad shorting a NeoPixel or something like that.

ladyada, 2018-11-14:
this would go in the neotrellis keypad wrapper library which isnt written yet

@jedgarpark please add a FAQ/Troubleshooting note to the NeoTrellis M4 guide for now till we write it!

jedgarpark, 2018-11-15:
will do

deanm1278, 2018-11-20:
this is fixed in neotrellis lib!

closed 2018-11-20 by deanm1278