adafruit/Adafruit_Keypad #2: error on build

garyocampo, 2018-11-17:
`Arduino: 1.8.7 (Mac OS X), TD: 1.44, Board: "Teensy 3.6, MIDI, 180 MHz, Faster, US English"

/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp: In member function 'void Adafruit_Keypad::tick()':
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:85:27: error: 'class RingBuffer' has no member named 'store_char'
                 _eventbuf.store_char(evt);
                           ^
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:86:27: error: 'class RingBuffer' has no member named 'store_char'
                 _eventbuf.store_char(*(_userKeymap + i));
                           ^
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:92:27: error: 'class RingBuffer' has no member named 'store_char'
                 _eventbuf.store_char(evt);
                           ^
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:93:27: error: 'class RingBuffer' has no member named 'store_char'
                 _eventbuf.store_char(*(_userKeymap + i));
                           ^
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp: In member function 'int Adafruit_Keypad::available()':
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:189:23: error: 'class RingBuffer' has no member named 'available'
     return (_eventbuf.available() >> 1);
                       ^
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp: In member function 'keypadEvent Adafruit_Keypad::read()':
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:201:29: error: 'class RingBuffer' has no member named 'read_char'
     k.bit.EVENT = _eventbuf.read_char();
                             ^
/Users/gabriel/Documents/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.cpp:202:27: error: 'class RingBuffer' has no member named 'read_char'
     k.bit.KEY = _eventbuf.read_char();
                           ^
Error compiling for board Teensy 3.6.
`
any ideas on how to fix?

garyocampo, 2018-11-17:
ok, so I was missing the following code on my libraries 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/RingBuffer.cpp
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/RingBuffer.h
I also had an issue with INPUTT_PULLDOWN not being a thing. apparently it's only for the zero. i'm using a teensy.
it's sorta working now, and by working I mean it uploads the code and it prints this very hopefull message:
`the (int)e.bit.KEY 1 the e.bit.EVENT 49
the (int)e.bit.KEY 0 the e.bit.EVENT 49
the (int)e.bit.KEY 1 the e.bit.EVENT 49
the (int)e.bit.KEY 0 the e.bit.EVENT 49
the (int)e.bit.KEY 1 the e.bit.EVENT 49
the (int)e.bit.KEY 0 the e.bit.EVENT 49
the (int)e.bit.KEY 0 the e.bit.EVENT 73
the (int)e.bit.KEY 0 the e.bit.EVENT 74
the (int)e.bit.KEY 1 the e.bit.EVENT 75
the (int)e.bit.KEY 1 the e.bit.EVENT 75
the (int)e.bit.KEY 1 the e.bit.EVENT 74
`
that's when I press the keys and also when the program begins it will print the following

`the (int)e.bit.KEY 1 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 9 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 17 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 25 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 33 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 41 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 49 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 57 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 65 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 73 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 81 the e.bit.EVENT 1
 pressed
the (int)e.bit.KEY 2 the e.bit.EVENT 1
 pressed
` 
so for some reason when i press a button e.bit.EVENT != KEY_JUST_PRESSED.

ladyada, 2018-11-18:
hiya yep right now this is very specialized for the Trellis M4 - not for generalized usage yet :) if you have fixes that you can contribute please submit a PR!

ladyada, 2018-11-26:
try the recent version for a PULLDOWN fix!

caternuson, 2019-06-28:
Please try the 1.0.3 version of the library when it becomes available.

caternuson, 2019-07-12:
Closing for now. Hopefully fixed by #4. Can reopen if needed.

closed 2019-07-12 by caternuson