adafruit/Adafruit_Keypad #3: Dependency on Ringbuffer.h not satisfied by Arduino Core

CristoXeuari, 2019-05-07:
Hi, my assumption is that Ringbuffer.h is supposed to be part of the arduino core. Perhaps this is missing in later version of arduino ide (I'm using 2:1.0.5). I would very much like to use this library but I'm unsure which supporting code i'm missing as there are several ringbuffer implementations available.
Thank you.

Code:
#include <Adafruit_Keypad.h>

Compilation:
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections /usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -MMD -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/eightanaloginputs -I/home/pi/Arduino/libraries/Adafruit_Keypad -I/home/pi/Arduino/libraries/Adafruit_Keypad /tmp/build7967350557655550057.tmp/sketch_may07a.cpp -o /tmp/build7967350557655550057.tmp/sketch_may07a.cpp.o 
-DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/eightanaloginputs -I/home/pi/Arduino/libraries/Adafruit_Keypad -I/home/pi/Arduino/libraries/Adafruit_Keypad /tmp/build7967350557655550057.tmp/sketch_may07a.cpp -o /tmp/build7967350557655550057.tmp/sketch_may07a.cpp.o 
In file included from sketch_may07a.ino:1:0:
In file included from sketch_may07a.ino:1:0:
/home/pi/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.h:5:24: fatal error: RingBuffer.h: No such file or directory
compilation terminated.
/home/pi/Arduino/libraries/Adafruit_Keypad/Adafruit_Keypad.h:5:24: fatal error: RingBuffer.h: No such file or directory
compilation terminated.

ladyada, 2019-05-07:
use latest ide, reopen if still issue

CristoXeuari, 2019-05-08:
I have the same problem using ide 1.8.9 which I believe is the latest version.

TheKitty, 2019-05-08:
Google: ringbuffer.h arduino

TheKitty, 2019-05-08:
If you google ringbuffer.h arduino there are results that may help. It is not code that is provided by Adafruit but part of the IDE. The Arduino forums would be another source of information.

caternuson, 2019-05-08:
What board are you using? It looks like RingBuffer is not implemented on all targets.

ladyada, 2019-05-08:
carter, dya want to try and generalize this? originally we wrote it for
neotrellis but its useful besides

caternuson, 2019-05-08:
@ladyada can take a look

@CristoXeuari still curious about your hardware setup

What's the non-NeoTrellis M4 use case for this? I'm probably just forgetting or overlooking something.

- [Trellis](https://www.adafruit.com/product/1616) - uses HT16K33
- [NeoTrellis](https://www.adafruit.com/product/3954) - uses seesaw
- [NeoTrellis M4](https://www.adafruit.com/product/3938) - has an M4 so can use this library

ladyada, 2019-05-09:
carter, you could include the ringbuffer file as a utiliy (under a new name)

CristoXeuari, 2019-05-17:
The board is a Nano made by Elegoo with an Atmega328p and CH340.
The keypad is made by Sodial, 3x4, uses 7 pins and, I think, membrane switches.

caternuson, 2019-06-28:
Please try the 1.0.3 version of the library when it becomes available.

caternuson, 2019-07-12:
Closing for now. Hopefully fixed by #4. Can reopen if needed.

closed 2019-07-12 by caternuson