adafruit/Adafruit_Keypad #6: Problem with the class

erzica, 2019-11-08:
Hi.
This is the error message on adafruit keypad library testing keypad_test arduino sketch.

Arduino:1.8.9 (Windows Store 1.8.21.0) (Windows 10), Tarjeta:"Arduino Due (Programming Port)"
D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp: In member function 'int Adafruit_Keypad::available()':
D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp:191:23: error: 'class RingBuffer' has no member named 'available'
     return (_eventbuf.available() >> 1);
                       ^
D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp: In member function 'keypadEvent Adafruit_Keypad::read()':

D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp:203:29: error: 'class RingBuffer' has no member named 'read_char'
     k.bit.EVENT = _eventbuf.read_char();
                             ^
D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp:204:27: error: 'class RingBuffer' has no member named 'read_char'
     k.bit.KEY = _eventbuf.read_char();
                           ^
D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp: In member function 'void Adafruit_Keypad::clear()':

D:\Users\Nacho\Documents\Arduino\libraries\Adafruit_Keypad\Adafruit_Keypad.cpp:217:12: error: 'class RingBuffer' has no member named 'clear'

  _eventbuf.clear();

            ^

exit status 1
Error compilando para la tarjeta Arduino Due (Programming Port).

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

caternuson, 2019-11-08:
What version of the library are you using?

erzica, 2019-11-08:
The last one.
It works with Arduino Mega, but it doesn’t work with Arduino Due.

caternuson, 2019-11-10:
Looks like not only do some cores have this and some do not, but for those that do - they have different implementations. Here's the one for Arduino SAM (Due):
https://github.com/arduino/ArduinoCore-sam/blob/master/cores/arduino/RingBuffer.h
and here's the one for Adafruit SAMD:
https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/RingBuffer.h

@ladyada We've dealt with this before in #3. The solution in #4 was to use the core if it exist and if not then provide the local version. Maybe this should just be changed to use *only* the local implementation?

ladyada, 2019-11-24:
hiya yep lets rename the internal version to like `Adafruit_Keypad_Ringbuffer` and only use the internal version.

caternuson, 2019-11-25:
@erzica Please try the 1.0.5 version of the library when it becomes available:
https://github.com/adafruit/Adafruit_Keypad/releases/tag/1.0.5

caternuson, 2019-12-04:
Closing due to lack of response. Can reopen if needed.

closed 2019-12-04 by caternuson