adafruit/Adafruit_Keypad #11: Not working on Arduino Due

dtman101, 2021-02-17:
- Arduino board:  Arduino Due

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

I've used this library on the Mega and Uno many times without issue, however the DUE does not seem to like it.
Is there something in the library that does not agree with the DUE? I've tried the library from Mark Stanley with luck, but would prefer to use the Adafruit library since it's already in so many of my programs.

`#include "Adafruit_Keypad.h"

//Keypad Setup

const byte ROWS = 4; //four rows
const byte COLS = 5; //four columns

char keys[ROWS][COLS] = { // Assign keys
  {'1', '2', '3', 'B', 'U'},
  {'4', '5', '6', 'X', 'D'},
  {'7', '8', '9', 'Y', 'V'},
  {'C', '0', 'E', 'P', 'S'}
};
byte rowPins[ROWS] = {43, 47, 49, 41}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {53, 51, 45, 39, 37}; //connect to the column pinouts of the keypad

//initialize an instance of class NewKeypad
Adafruit_Keypad customKeypad = Adafruit_Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
  Serial.begin(115200);
  customKeypad.begin();
  Serial.println("init");

}

void loop() {
  // put your main code here, to run repeatedly:
  customKeypad.tick();

  while(customKeypad.available()){
    keypadEvent e = customKeypad.read();
    Serial.print((char)e.bit.KEY);
    if(e.bit.EVENT == KEY_JUST_PRESSED) Serial.println(" pressed");
    else if(e.bit.EVENT == KEY_JUST_RELEASED) Serial.println(" released");
  }

  delay(10);
}`

ladyada, 2021-02-17:
Due has a lot of issues, not sure what it is, we dont test our hardware with the Due

DukeIsCreating, 2025-03-30:
This is due to a conflict in the use of the SERIAL_BUFFER_SIZE define, which is already defined by the Arduino's built-in system libraries RingBuffer.h.

Renaming the SERIAL_BUFFER_SIZE define in Adafruit_Keypad_RingBuffer.h and updating all references to it within that file allowed at least the Adafruit_Keypad_Ringbuffer to work for me without crashing the Due. I can't speak for the matrixed keypad code though because I had actually re-written the Adafruit_Keypad class to work with a non-matrixed keypad I had lying around.